Cap-go/capacitor-camera-preview #2: Need Video Recording for iOS.

Saqib92, 2022-12-04:
are we getting video recording in iOS?

Saqib92, 2022-12-05:
> Yes as in android

any ETA?

riderx, 2022-12-06:
Sorry my answer was "yes that work like android already", i just saw the doc was not updated, IOS is implemented

Saqib92, 2022-12-06:
> Sorry my answer was "yes that work like android already", i just saw the doc was not updated, IOS is implemented

Okay i will test it and let you know.

Saqib92, 2022-12-14:
[image]
Video Recording not working.

riderx, 2022-12-14:
Seems the definition was broken, after PR merge, i fixxed it

Saqib92, 2022-12-14:
Still showing same error. Please update npm package so I can install/test new version.

riderx, 2023-02-24:
sorry released it, but forgot to tell you, did you checked ?

Saqib92, 2023-02-24:
No, Will check and let you know. thanks,

closed 2024-05-03 by riderx