Cap-go/capacitor-camera-preview #24: Can't get the same size with the preview and captureSample

FernetB, 2023-02-07:
I can't match 100% the preview size with the capturaSame size (with capture is even worse)

Currently I start the preview with only 
```js
 position: "rear",
 toBack: true,
```
and captureSame method with nothing

You can see the difference in the head of the bottle
(<img /> tag, that has the picture is full screen, is not cropped) 
Preview:
[image]

Photo:
[image]

FernetB, 2023-02-13:
@riderx is this repo still mantained?

riderx, 2023-02-13:
yes i was just in vacation,

thanks for the report, i'm not sure to understand your issue, the photo taken is not the same size than the preview ?

FernetB, 2023-02-14:
Oh sorry for the trouble!

There are 2 problems:

 - Capture: this take a native picture, and depending on the phone, sometimes the picture is taken with another lens, and therefore the photo does not match what you were seeing on the preview
 - CaptureSample: this take a frame from the preview, but for me does not match exactly the preview. In the two bottle's pictures you can see that the bottle cap is visible on the first picture (the preview) and is sliced on the second picture (the captureSample)

I hope I have explained myself correctly

kenjoesavandal, 2023-05-25:
@FernetB did you find a fix? I have the same issue. :(

riderx, 2024-05-03:
Hey guys, i finally took a look thanks for reporting it, it should be fixed now !

riderx, 2024-05-03:
i didn't touched CaptureSample, but fixed Capture

closed 2024-05-11 by riderx