Cap-go/capacitor-camera-preview #35: Video recording on android crashed the app.

Saqib92, 2023-04-14:
startRecording on android crashed the app. 

Demo repo:
[Git repo](https://github.com/Saqib92/cam-preview)

fedee13, 2023-04-24:
If it's of any help, I have the same issue and this is the log from Android Studio

```
D/Preview: screen is rotated 0deg from natural
D/Preview: back camera is oriented -90deg from natural
D/Preview: need to rotate preview 90deg
D/CameraActivity: cameraCurrentlyLocked:0
I/Capacitor/Console: File: http://192.168.1.2:8200/ - Line 230 - Msg: undefined
D/Preview: optimal preview size: w: 1920 h: 1440
D/Preview: optimal preview size: w: 1920 h: 1440
D/Preview: center horizontally
D/layout: left:-225
D/layout: top:0
D/layout: right:1305
D/layout: bottom:2040
D/Preview: optimal preview size: w: 176 h: 144
I/chatty: uid=10674(com.demo.camera) identical 1 line
D/Preview: optimal preview size: w: 176 h: 144
D/SurfaceView: UPDATE null, mIsCastMode = false
D/Preview: optimal preview size: w: 1920 h: 1440
W/Looper: PerfMonitor doFrame : time=311ms vsyncFrame=41776824 latency=249ms procState=2 historyMsgCount=2 (msgIndex=2 wall=232ms seq=184 running=167ms runnable=5ms late=26ms h=android.os.Handler c=android.app.FragmentManagerImpl$1)
D/Preview: optimal preview size: w: 1920 h: 1440
D/Preview: optimal preview size: w: 1920 h: 1440
W/Looper: PerfMonitor doFrame : time=98ms vsyncFrame=41776826 latency=314ms procState=2 historyMsgCount=3 (msgIndex=1 wall=311ms seq=185 running=106ms runnable=9ms late=249ms h=android.view.Choreographer$FrameHandler c=android.view.Choreographer$FrameDisplayEventReceiver)
V/Capacitor/Plugin: To native (Capacitor plugin): callbackId: 43867362, pluginId: CameraPreview, methodName: startRecordVideo
V/Capacitor: callback: 43867362, pluginId: CameraPreview, methodName: startRecordVideo, methodData: {"position":"rear","width":360,"height":780}
D/CameraActivity: CameraPreview startRecord camera: rear width: 360, height: 780, quality: 70
E/libc: Access denied finding property "camera.aux.packagelist"
W/CameraActivity: ************orientationHint ***********= 90
D/AndroidRuntime: Shutting down VM
E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.demo.camera, PID: 10424
    java.lang.RuntimeException: setAudioSource failed.
        at android.media.MediaRecorder.setAudioSource(Native Method)
        at com.ahm.capacitor.camera.preview.CameraActivity.startRecord(CameraActivity.java:865)
        at com.ahm.capacitor.camera.preview.CameraPreview$2.run(CameraPreview.java:238)
        at android.os.Handler.handleCallback(Handler.java:938)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:236)
        at android.app.ActivityThread.main(ActivityThread.java:8061)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:656)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:967)
```
I even tried to set the following permissions in AndroidManifest but it only ask for the camera one:
```
    <uses-permission android:name="android.permission.CAMERA" />
    <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE"/>
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.RECORD_AUDIO" />
```

fedee13, 2023-04-26:
This is how I open/record:
```ts
async function openCamera() {
  const cameraPreviewOptions: CameraPreviewOptions = {
    parent: "camera-preview",
    position: "rear",
    width: window.screen.width,
    height: window.screen.height - 100,
    enableZoom: true
  };

  await CameraPreview.start(cameraPreviewOptions);
}
```
```ts
async function recordVideo() {
  const cameraPreviewOptions: CameraPreviewOptions = {
    position: "rear",
    width: window.screen.width,
    height: window.screen.height
  };

  await CameraPreview.startRecordVideo(cameraPreviewOptions);
}
```
The app crashes at `await CameraPreview.startRecordVideo(cameraPreviewOptions);`

riderx, 2024-05-03:
Hey guys thanks for the feedback and sorry for the delay i wasn't really available for this.
I managed to find the issue and fixed it !
it now works and also have fixed definition where you can get the url of the video on ios as well.
Please re open if you have still the issue

closed 2024-05-03 by riderx