Cap-go/capacitor-camera-preview #36: stopRecordVideo does not return video path

Reizar, 2023-04-18:
Hi,

Thanks for the hard work on updating this library.

I've noticed that the actual types / results for the `stopRecordVideo` function doesn't match up with the docs. The actual return type is Promise<void>. So theres no file path of the video once its captured. 

An update for this would be greatly appreciated as at the moment the video recording can't be used.

Thanks!

riderx, 2024-05-03:
hey sorry to have missed your query i'm not sure to understand, you shouldn't get any path as result.
nothing in the code return a path

riderx, 2024-05-03:
ok i got it, the videoUrl was returned on android, it's now in the definition and i also added the feature on ios

closed 2024-05-03 by riderx