Cap-go/capacitor-camera-preview #61: Migrate Android Camera to CameraX of Camera2

riderx, 2024-05-03:
The plugin uses a deprecated lib to do its job, it could be nice to migrate it to the newest lib.
Any of the 2 options is great if All features work properly

riderx, 2024-05-03:
/bounty $100

neo773, 2024-05-03:
/attempt #61

riderx, 2024-05-03:
@neo773 thanks for the reactivity, i updated the demo to make your life easier all features are available

amaanlari, 2024-05-13:
/attempt #61

asr2003, 2024-07-05:
/attempt #61

dpa99c, 2024-07-06:
as @forgr-owner mentions above, I have implemented the Camera 2 API in https://github.com/capacitor-community/camera-preview/pull/339 - not sure how similar this repo is as don't have time to look right now but feel free to use as doesn't seem like my PR will be merged

asr2003, 2024-07-06:
Hi @dpa99c ,

I have taken your implementation as reference and implemented here https://github.com/Cap-go/camera-preview/pull/87. Can you have a look at it and review them?

riderx, 2024-07-06:
@dpa99c This repo is a fork of the original one you created the PR.
Since I'm personally banned from Capacitor or I couldn't Fork it.
I just fixed, maintained and merged a lot of PR, so migrating is probably easy.
@asr2003 did a PR but cannot run Gradle on his machine.
So if you are up to helping him to review his implementation of your code, I can double the bounty and split it with both of you.
Clients of Capgo are looking for this change too

asr2003, 2024-07-06:
And to be in same page we are implementing the CameraX which is already merged in the capacitor successfully. 
@riderx If you double up the bounty(or tip in the PR) I will take help of my friend PC to install gradle in his MAC and test changes and will add the video of the changes in the PR

dpa99c, 2024-07-07:
> Hi @dpa99c ,
> 
> I have taken implementation of CameraX in capacitor as reference and implemented here #87. Can you have a look at it and review them?
> 
> @dpa99c also I have small queries on your work, Can i get your discord id?

@asr2003 it seems your PR is a migration to CameraX API which is a replacement for legacy Camera API, which are both high-level abstractions of the camera hardware.

Whereas my PR directly uses the Camera2 API, which provides low-level access to the camera hardware. 
In addition to fixing some outstanding issues, it opens the way for future enhancements such as selecting a specific lens to use when the logical camera consists of multiple lenses.

I'm not really familiar with the CameraX API (nor am I really interested in getting familiar with it) so I'm not really able to review your PR.
I believe it's possible to use Camera2 API functionality in a CameraX implementation via the [Camera2Interop](https://developer.android.com/reference/androidx/camera/camera2/interop/Camera2Interop) class but I don't have familiarity with this.

@riderx it's up to you whether you want this plugin to use Camera2 API directly or CameraX in conjuction with Camera2 via Camera2Interop.
I don't have the time to test or implement the CameraX version but you are welcome to use my PR above for the Camera2 implementation.

riderx, 2024-07-07:
Thanks for looking at it @dpa99c,
I believe Camera 2 is a better option as it allows lower level access and more flexibility.
 @asr2003 if you manage to have a computer who works with Gradle, I can sponsor it and if @dpa99c you can just review the implementation I can still support it for that and will be delighted

forgr-owner, 2024-07-07:
@neo773 If you want to check the PR of @dpa99c as well

asr2003, 2024-07-07:
@forgr-owner I am more interested to complete this issue :)

FYI, @neo773 has cancelled his attempt

asr2003, 2024-07-07:
@riderx Do my PR need to be update to integrate Camera2 instead the CameraX?

riderx, 2024-07-07:
@asr2003 yes,
I'm glad you are motivated to complete the task, and I'm trying to find a solution that you could do while I get the expected result.
The PR you made in Capgo org were 2 time not tested and not working.
Without the help of someone I trust, I'm not comfortable letting do it on your own, as the chances are high to require me to work on it too.
So I proposed to @neo773 to help you because he proved his capacity to deliver well-tested code, multiple time.

asr2003, 2024-07-07:
@riderx Thanks! I will follow up this issue with the @neo773 and complete this issue very soon :)

dpa99c, 2024-07-07:
If you want to validate my [Camera2 API implementation](https://github.com/dpa99c/capacitor-camera-preview), you can use my test app project which pulls it in:
https://github.com/dpa99c/capacitor-camera-preview-test

asr2003, 2024-07-08:
@riderx Waiting for the @neo773 response

riderx, 2024-07-08:
@asr2003 I got an answer in private @neo773 will be able to review your code, but please do build on your side before and test as well

riderx, 2024-07-08:
To make a PR valid you need to provide a demo video, as it say in the instruction, so i will expect that before any review

asr2003, 2024-07-09:
Sure @riderx

rohit9625, 2024-12-13:
Is anyone still working on this issue?

riderx, 2024-12-13:
No the bounty is closed we will look for a lib instead of doing all manually

riderx, 2024-12-13:
close in favor of https://github.com/Cap-go/camera-preview/issues/147

closed 2024-12-13 by riderx

rohit9625, 2024-12-13:
> No the bounty is closed we will look for a lib instead of doing all manually

Okay, No problem :)