Cap-go/capacitor-camera-preview #75: Feat: RECORD_AUDIO Permission shouldn't be required

qliqdev, 2024-06-07:
Hi. What if we use camera preview only for taking photos.

Is it possible to dynamicaly check and request `RECORD_AUDIO` permission even if client calls `startRecordVideo` function?

selected-pixel-jameson, 2024-06-12:
Ya. This needs to be removed. This is a huge issue. It's really screwed me with a recent deployment as I didn't realize this permission got added and it's most definitely not needed just to take a picture.

riderx, 2024-09-03:
We will fix it thanks for the feedback

riderx, 2024-09-03:
@qliqdev did you set `disableAudio` to true when you call the plugin?

riderx, 2024-12-09:
We published a fix for Android, it was missing on android please try [6.5.0](https://github.com/Cap-go/camera-preview/tree/refs/tags/6.5.0)

closed 2024-12-09 by riderx

qliqdev, 2024-12-09:
> We published a fix for Android, it was missing on android please try [6.5.0](https://github.com/Cap-go/camera-preview/tree/refs/tags/6.5.0)

Thx a lot