Cap-go/capacitor-camera-preview #84: How to convert the path to a blob or base64 on iOS?

wijzijnweb, 2024-06-25:
On iOS I cant seem to fetch or read the file using fetch or the capacitor filesystem.
File is always "Not found".

On Android it returns: result.videoFilePath
On iOS it returns: result.videoUrl (with file://)

This can't be fetched since it only supports http.
The filesystem also says it's not found.

Any idea on how to solve this?
I've tried:

```
 const result = await Filesystem.readFile({
            path: filePath,
            directory: FilesystemDirectory.Library, 
            encoding: Encoding.BASE64
        });
```

A normal `fetch(result.videoUrl)` won't work.

wijzijnweb, 2024-06-25:
```
func captureVideo() throws {
        guard let captureSession = self.captureSession, captureSession.isRunning else {
            throw CameraControllerError.captureSessionIsMissing
        }
        let path = FileManager.default.urls(for: .cachesDirectory, in: .userDomainMask)[0]
        let identifier = UUID()
        let randomIdentifier = identifier.uuidString.replacingOccurrences(of: "-", with: "")
        let finalIdentifier = String(randomIdentifier.prefix(8))
        let fileName="cpcp_video_"+finalIdentifier+".mp4"
        let fileUrl = path.appendingPathComponent(fileName)
        try? FileManager.default.removeItem(at: fileUrl)
        // Start recording video
        // ...
        // Save the file URL for later use
        self.videoFileURL = fileUrl
    }
```

Not sure, but is the video deleted on capture?

riderx, 2024-12-09:
@WcaleNieWolny did you say this kind of issue ?

WcaleNieWolny, 2024-12-10:
@riderx no, but I did write an app that uploaded the recorded video into s3.

```ts
  async function record() {
    if (!isRecording.value) {
      isRecording.value = true
      await CameraPreview.startRecordVideo({ storeToFile: true })
    } else {
      isRecording.value = false
      const fileUrl = await CameraPreview.stopRecordVideo()
      console.log(fileUrl.videoFilePath)
      // const contents = await Filesystem.readFile({
      //   path: fileUrl.videoFilePath,
      // });
      // console.log(contents.data.length)

      const randomId = makeid(8)
      console.log(`Attempt to upload ${randomId}`)

      const urlReq = await fetch(`https://[redacted]/${randomId}.mp4`, {
        headers: {
          'Authorization': 'Bearer [removed]'
        },
        method: 'GET'
      })

      const uploadUrl = (await urlReq.json() as any).url as string
      console.log('upload url', uploadUrl)

      const { id } = await Uploader.startUpload({
        filePath: fileUrl.videoFilePath,
        serverUrl: uploadUrl,
        method: 'PUT',
        // parameters: fields,
        headers: {},
        // notificationTitle: 'Uploading to S3'
      });

      console.log(`Plugin returned ${id}`)

    }
  }
  ```
  
  
[image]

From what the user shared, I think it's an issue on the user's end. The proper way would be to read `res.videoFilePath`, but the user uses `res.videoUrl` which is undefined.

```ts
const base64 = await Filesystem.readFile({
        path: res.videoUrl // <--- here, please put "videoFilePath"
});
```

closed 2024-12-10 by WcaleNieWolny