Cap-go/capacitor-camera-preview #107: Throwng error `capture()` and `startRecording()` in android 14

Saqib92, 2024-09-09:
Throwing error when trying any method:

`{message: 'Camera is not running'}`

```
async recordVid() {
    try {
      let opt: CameraPreviewOptions = {
        position: 'front',
        width: window.screen.width,
        height: window.screen.height
      }
      await CameraPreview.startRecordVideo(opt)
      
      await new Promise(resolve => setTimeout(resolve, 5000))

      const fileUrl = await CameraPreview.stopRecordVideo()
      console.log(fileUrl)
      
    }
    catch (err) {
      console.log(err)
    }

  }
```

Getting this error:

[image]

Added permissions in manifest:

```
<uses-permission android:name="android.permission.CAMERA" />
<uses-permission android:name="android.permission.RECORD_AUDIO" />
<uses-permission android:name="android.permission.MODIFY_AUDIO_SETTINGS" />

<uses-permission android:name="android.permission.READ_MEDIA_VIDEO" />
<uses-permission android:name="android.permission.READ_MEDIA_IMAGES" />
<uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
<uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
```

WcaleNieWolny, 2024-11-11:
@Saqib92 the code you shared does not correspond to the error you shared. In either way, I would recommend doing the following inside of the component you use for the camera:

```ts
// vue.js specific code
onUnmounted(async () => {
  await CameraPreview.stop() // <-- important, stop the camera when you unmount
})

onMounted(() => {
  await CameraPreview.start({
    toBack: true,
    parent: 'cam-prev'
  })
})
```

Personally, I was experiencing a similar errors to yours, but this `CameraPreview.stop` call helped me resolve it

Saqib92, 2024-11-11:
> @Saqib92 the code you shared does not correspond to the error you shared. In either way, I would recommend doing the following inside of the component you use for the camera:
>
> ```ts
> // vue.js [...]

Yes, i found this `stop()` method helpful.

closed 2024-11-11 by Saqib92