Cap-go/capacitor-camera-preview #136: Video Recording crash on front camera

Saqib92, 2024-11-16:
Plugin in working fine in android 14 Both rare and Front but Crashing on android 13 and below on front cam:
logcat:

```
2024-11-16 16:18:13.236  9093-9093  VRI[MainActivity]       com.wizdiary.app                     D  handleWindowFocusChanged mWindowFocusChanged false mUpcomingWindowFocus true mAdded true
2024-11-16 16:18:13.236  9093-9093  AutofillManager         com.wizdiary.app                     V  requestHideFillUi(null): anchor = null
2024-11-16 16:18:13.257  9093-9093  VRI[MainActivity]       com.wizdiary.app                     D  handleWindowFocusChanged mWindowFocusChanged false mUpcomingWindowFocus true mAdded true
2024-11-16 16:18:13.273  9093-9093  VRI[MainActivity]       com.wizdiary.app                     D  handleWindowFocusChanged mWindowFocusChanged false mUpcomingWindowFocus true mAdded true
2024-11-16 16:18:13.291  9093-9093  VRI[MainActivity]       com.wizdiary.app                     D  handleWindowFocusChanged mWindowFocusChanged false mUpcomingWindowFocus true mAdded true
2024-11-16 16:18:13.307  9093-9093  VRI[MainActivity]       com.wizdiary.app                     D  handleWindowFocusChanged mWindowFocusChanged false mUpcomingWindowFocus true mAdded true
2024-11-16 16:18:13.327  9093-9093  VRI[MainActivity]       com.wizdiary.app                     D  handleWindowFocusChanged mWindowFocusChanged false mUpcomingWindowFocus true mAdded true
2024-11-16 16:18:13.339  9093-9093  Capacitor/Plugin        com.wizdiary.app                     V  To native (Capacitor plugin): callbackId: 12428589, pluginId: CameraPreview, methodName: startRecordVideo
2024-11-16 16:18:13.340  9093-9093  Capacitor               com.wizdiary.app                     V  callback: 12428589, pluginId: CameraPreview, methodName: startRecordVideo, methodData: {"storeToFile":true}
2024-11-16 16:18:13.343  9093-9093  CameraActivity          com.wizdiary.app                     D  CameraPreview startRecord camera: front width: 0, height: 0, quality: 70
2024-11-16 16:18:13.344  9093-9093  libc                    com.wizdiary.app                     W  Access denied finding property "vendor.camera.aux.packagelist"
2024-11-16 16:18:13.347  9093-9093  CameraActivity          com.wizdiary.app                     W  ************orientationHint ***********= 90
2024-11-16 16:18:13.348  9093-9093  MediaRecorder           com.wizdiary.app                     D  constructor+ 0xb400007bfca21c80
2024-11-16 16:18:13.349  9093-9093  MediaRecorder           com.wizdiary.app                     D  constructor- 0xb400007bfca21c80
2024-11-16 16:18:13.349  9093-9093  CheckPermission         com.wizdiary.app                     E  media_record-code= 24
2024-11-16 16:18:13.350  9093-9093  CheckPermission         com.wizdiary.app                     E  media_record-code= 22
2024-11-16 16:18:13.350  9093-9093  MediaRecorder           com.wizdiary.app                     D  init+ 0xb400007bfca21c80
2024-11-16 16:18:13.350  9093-9093  CheckPermission         com.wizdiary.app                     E  media_record-code= 2
2024-11-16 16:18:13.350  9093-9093  CheckPermission         com.wizdiary.app                     E  media_record-code= 23
2024-11-16 16:18:13.363  9093-9093  MediaRecorder           com.wizdiary.app                     D  init- 0xb400007bfca21c80
2024-11-16 16:18:13.363  9093-9093  CheckPermission         com.wizdiary.app                     E  media_record-code= 12
2024-11-16 16:18:13.364  9093-9093  MediaRecorderJNI        com.wizdiary.app                     D  setVideoSource(1)
2024-11-16 16:18:13.364  9093-9093  MediaRecorderJNI        com.wizdiary.app                     D  getMediaRecorder-
2024-11-16 16:18:13.364  9093-9093  CheckPermission         com.wizdiary.app                     E  media_record-code= 11
2024-11-16 16:18:13.365  9093-9093  MediaRecorderJNI        com.wizdiary.app                     D  setVideoSource-
2024-11-16 16:18:13.365  9093-9093  CheckPermission         com.wizdiary.app                     E  media_record-code= 13
2024-11-16 16:18:13.365  9093-9093  CheckPermission         com.wizdiary.app                     E  media_record-code= 19
2024-11-16 16:18:13.365  9093-9093  CheckPermission         com.wizdiary.app                     E  media_record-code= 18
2024-11-16 16:18:13.365  9093-9093  CheckPermission         com.wizdiary.app                     E  media_record-code= 20
2024-11-16 16:18:13.365  9093-9093  CheckPermission         com.wizdiary.app                     E  media_record-code= 14
2024-11-16 16:18:13.365  9093-9093  CheckPermission         com.wizdiary.app                     E  media_record-code= 20
2024-11-16 16:18:13.365  9093-9093  CheckPermission         com.wizdiary.app                     E  media_record-code= 20
2024-11-16 16:18:13.366  9093-9093  CheckPermission         com.wizdiary.app                     E  media_record-code= 20
2024-11-16 16:18:13.366  9093-9093  CheckPermission         com.wizdiary.app                     E  media_record-code= 15
2024-11-16 16:18:13.366  9093-9093  CheckPermission         com.wizdiary.app                     E  media_record-code= 20
2024-11-16 16:18:13.366  9093-9093  CheckPermission         com.wizdiary.app                     E  media_record-code= 20
2024-11-16 16:18:13.367  9093-9093  CheckPermission         com.wizdiary.app                     E  media_record-code= 16
2024-11-16 16:18:13.367  9093-9093  MediaRecorder           com.wizdiary.app                     D  prepare+ 0xb400007bfca21c80
2024-11-16 16:18:13.367  9093-9093  CheckPermission         com.wizdiary.app                     E  media_record-code= 9
2024-11-16 16:18:13.367  9093-9093  MediaRecorder           com.wizdiary.app                     D  prepare- 0xb400007bfca21c80
2024-11-16 16:18:13.367  9093-9093  CameraActivity          com.wizdiary.app                     D  Starting recording
2024-11-16 16:18:13.367  9093-9093  MediaRecorder           com.wizdiary.app                     D  start+ 0xb400007bfca21c80
2024-11-16 16:18:13.367  9093-9093  CheckPermission         com.wizdiary.app                     E  media_record-code= 8
2024-11-16 16:18:13.371  9093-9093  MediaRecorder           com.wizdiary.app                     E  start failed: -19 0xb400007bfca21c80
2024-11-16 16:18:13.371  9093-9093  AndroidRuntime          com.wizdiary.app                     D  Shutting down VM
2024-11-16 16:18:13.372  9093-9093  AndroidRuntime          com.wizdiary.app                     E  FATAL EXCEPTION: main
                                                                                                    Process: com.wizdiary.app, PID: 9093
                                                                                                    java.lang.RuntimeException: start failed.
                                                                                                    	at android.media.MediaRecorder.start(Native Method)
                                                                                                    	at com.ahm.capacitor.camera.preview.CameraActivity.startRecord(CameraActivity.java:1129)
                                                                                                    	at com.ahm.capacitor.camera.preview.CameraPreview$2.run(CameraPreview.java:275)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:942)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:240)
                                                                                                    	at android.os.Looper.loop(Looper.java:351)
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:8423)
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method)
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:584)
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1013)
2024-11-16 16:18:13.424  9093-9093  Process                 com.wizdiary.app                     I  Sending signal. PID: 9093 SIG: 9
---------------------------- PROCESS ENDED (9093) for package com.wizdiary.app ----------------------------

```

Saqib92, 2024-12-10:
```
async openCamera() {
    let opt: CameraPreviewOptions = {
      position: 'rear',
      toBack: true
    }
    await CameraPreview.start(opt);
  }

  async flip() {
    await CameraPreview.flip();
  }

  async startRecording() {
    this.isRecording = true;
    await CameraPreview.startRecordVideo({ storeToFile: true });
  }
```

Issue come after i flip and start recording.

riderx, 2025-02-09:
can you ttest the v7 i think i fixed it

Saqib92, 2025-02-10:
Okay i will check in coming week. currently busy in other projects. Thank you