Cap-go/capacitor-camera-preview #146: Bug: 'disableAudio' true, still requests for audio permission on Android

TobiasHesselinkTopicus, 2024-12-05:
# Bug Report: Audio Permission Prompt Despite `disableAudio` Set to `true`

## Description

When setting the `disableAudio` property to `true` in the `CameraPreviewOptions`, the application still prompts for audio permissions on Android devices. This issue does not occur on iOS devices, where no audio permission is requested.

## Steps to Reproduce

1. Use the following code snippet to trigger the camera:

    ```typescript
    const cameraPreviewOptions: CameraPreviewOptions = {
        position: 'rear',
        parent: 'cameraPreview',
        toBack: true,
        disableAudio: true,
        y: offsetY,
        height: cameraPreviewHeight
    };
    return from(CameraPreview.start(cameraPreviewOptions));
    ```

2. Run the application on an Android device.
3. Observe that the app requests audio permissions even though `disableAudio` is set to `true`.

## Expected Behavior

When `disableAudio` is set to `true`, the app should not request audio permissions on Android devices.

## Actual Behavior

The app prompts for audio permissions on Android, despite `disableAudio` being set to `true`.

## Additional Context

- The following permission has been added to the Android manifest to prevent crashes when audio recording is attempted:

    ```xml
    <uses-permission android:name="android.permission.MODIFY_AUDIO_SETTINGS" />
    ```

- This issue only affects Android devices. iOS devices behave as expected and do not request audio permissions.

## Environment

| Package                  | Version |
|--------------------------|---------|
| @capgo/camera-preview    | &6.2.31 |
| @capacitor/android       | ^6.1.2  |
| @capacitor/app           | ^6.0.1  |
| @capacitor/camera        | ^6.0.2  |
| @capacitor/core          | ^6.1.2  |
| @capacitor/device        | ^6.0.1  |
| @capacitor/ios           | ^6.1.2  |
| @angular/animations      | 17.3.9  |
| @angular/cdk             | 17.3.9  |
| @angular/common          | 17.3.9  |
| @angular/compiler        | 17.3.9  |
| @angular/core            | 17.3.9  |
| @angular/fire            | ^17.1.0 |
| @angular/forms           | 17.3.9  |
| @angular/material        | 17.3.9  |
| @angular/platform-browser| 17.3.9  |
| @angular/platform-browser-dynamic | 17.3.9 |
| @angular/router          | 17.3.9  |
| @ionic/angular           | ^7.8.6  |

## Possible Solution

Investigate why the audio permission is being requested despite the `disableAudio` flag and provide a fix or workaround to prevent the permission prompt on Android.

riderx, 2024-12-09:
We published a fix for Android, it was missing on it please try [6.5.0](https://github.com/Cap-go/camera-preview/tree/refs/tags/6.5.0)

TobiasHesselinkTopicus, 2024-12-10:
@riderx Thanks for the fast response. I will take a look into this tomorrow. Keep you updated.

riderx, 2024-12-12:
You are welcome :)