Cap-go/capacitor-camera-preview #148: Video Capture Broken in v6.5.10 on Android

geffersonFerraz, 2024-12-14:
I believe the commit e06eaf9a97f463dd1a60b051a66ddb83ec73b62c is causing my issues.
I switched to the oldest tag, 6.3.35, and everything is working fine now.

```
2024-12-14 13:38:32.166  5984-5984  Capacitor               [my-app].app                      D  CliCKED: 1
2024-12-14 13:38:32.263  5984-5984  Capacitor/Plugin        [my-app].app                      V  To native (Capacitor plugin): callbackId: 125430410, pluginId: CameraPreview, methodName: start
2024-12-14 13:38:32.266  5984-5984  Capacitor               [my-app].app                      V  callback: 125430410, pluginId: CameraPreview, methodName: start, methodData: {"position":"rear","parent":"cameraPreview","className":"camera-preview","toBack":false,"width":412,"height":915,"enableZoom":true,"lockAndroidOrientation":true,"disableAudio":true}
2024-12-14 13:38:32.574  5984-5984  Preview                 [my-app].app                      D  screen is rotated 0deg from natural
2024-12-14 13:38:32.575  5984-5984  Preview                 [my-app].app                      D  back camera is oriented -90deg from natural
2024-12-14 13:38:32.575  5984-5984  Preview                 [my-app].app                      D  need to rotate preview 90deg
2024-12-14 13:38:32.596  5984-5984  CameraActivity          [my-app].app                      D  cameraCurrentlyLocked:0
2024-12-14 13:38:32.604  5984-5984  Capacitor/Console       [my-app].app                      I  File:  - Line 353 - Msg: undefined
2024-12-14 13:38:32.634  5984-5984  Preview                 [my-app].app                      D  optimal preview size: w: 1600 h: 720
2024-12-14 13:38:32.639  5984-5984  Preview                 [my-app].app                      D  optimal preview size: w: 1600 h: 720
2024-12-14 13:38:32.677  5984-5984  Preview                 [my-app].app                      D  center vertically
2024-12-14 13:38:32.679  5984-5984  layout                  [my-app].app                      D  left:0
2024-12-14 13:38:32.679  5984-5984  layout                  [my-app].app                      D  top:0
2024-12-14 13:38:32.679  5984-5984  layout                  [my-app].app                      D  right:721
2024-12-14 13:38:32.679  5984-5984  layout                  [my-app].app                      D  bottom:1601
2024-12-14 13:38:32.696  5984-5984  Preview                 [my-app].app                      D  optimal preview size: w: 176 h: 144
2024-12-14 13:38:32.701  5984-5984  chatty                  [my-app].app                      I  uid=10279([my-app].app) identical 1 line
2024-12-14 13:38:32.710  5984-5984  Preview                 [my-app].app                      D  optimal preview size: w: 176 h: 144
2024-12-14 13:38:32.725  5984-5984  Preview                 [my-app].app                      D  Preview size in surfaceChanged: 176x144
2024-12-14 13:38:32.731  5984-5984  Preview                 [my-app].app                      D  optimal preview size: w: 1600 h: 720
2024-12-14 13:38:32.935  5984-5984  Capacitor/Plugin        [my-app].app                      V  To native (Capacitor plugin): callbackId: 125430411, pluginId: CameraPreview, methodName: startRecordVideo
2024-12-14 13:38:32.938  5984-5984  Capacitor               [my-app].app                      V  callback: 125430411, pluginId: CameraPreview, methodName: startRecordVideo, methodData: {"position":"rear","parent":"cameraPreview","className":"camera-preview","toBack":false,"width":412,"height":915,"enableZoom":true,"lockAndroidOrientation":true,"disableAudio":true,"storeToFile":true}
2024-12-14 13:38:32.957  5984-5984  Preview                 [my-app].app                      D  optimal preview size: w: 1600 h: 720
2024-12-14 13:38:32.962  5984-5984  Preview                 [my-app].app                      D  optimal preview size: w: 1600 h: 720
2024-12-14 13:38:32.989  5984-5984  CameraActivity          [my-app].app                      D  CameraPreview startRecord camera: rear width: 412, height: 915, quality: 70
2024-12-14 13:38:32.994  5984-5984  CameraActivity          [my-app].app                      W  ************orientationHint ***********= 90
2024-12-14 13:38:33.003  5984-5984  MediaRecorder           [my-app].app                      E  try to set the audio encoder without setting the audio source first
2024-12-14 13:38:33.004  5984-5984  AndroidRuntime          [my-app].app                      D  Shutting down VM
2024-12-14 13:38:33.010  5984-5984  AndroidRuntime          [my-app].app                      E  FATAL EXCEPTION: main
                                                                                                    Process: [my-app].app, PID: 5984
                                                                                                    java.lang.IllegalStateException
                                                                                                    	at android.media.MediaRecorder.setAudioEncoder(Native Method)
                                                                                                    	at android.media.MediaRecorder.setProfile(MediaRecorder.java:677)
                                                                                                    	at com.ahm.capacitor.camera.preview.CameraActivity.startRecord(CameraActivity.java:1122)
                                                                                                    	at com.ahm.capacitor.camera.preview.CameraPreview$2.run(CameraPreview.java:354)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:938)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                                                    	at android.os.Looper.loop(Looper.java:250)
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:7806)
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method)
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:958)
2024-12-14 13:38:33.060  5984-5984  Process                 [my-app].app                      I  Sending signal. PID: 5984 SIG: 9
```

zachtaylor, 2025-01-14:
I am experiencing the same crash from v6.5.0 (there is no v6.4.0 on npm). It seems the commit `e06eaf9` moved `setAudioSource` to after `setProfile`

riderx, 2025-02-08:
Sorry for that merged and fixed by @zachtaylor

closed 2025-02-08 by riderx