Cap-go/capacitor-camera-preview #172: iOS: whole app crashes after opening and closing the camera

Albermonte, 2025-02-05:
iOS: 18.3
@capgo/camera-preview:  6.5.24

When calling open and close before the camera has been initialized the app crashes throwing a `Fatal error`:
```
Thread 1: Fatal error: Unexpectedly found nil while implicitly unwrapping an Optional value
```

Error line: [Plugin.swift#L247](https://github.com/Cap-go/camera-preview/blob/f7c6da7617730df925fa5a946612d868b3e0ceda/ios/Plugin/Plugin.swift#L247)

Reproduction example:
```
// Yes, no await
CameraPreview.start({ enableHighResolution: true, disableAudio: true, cameraMode: true })
CameraPreview.stop()
```

Only happening on iOS, tested on Android and working

riderx, 2025-02-09:
Thanks for the report, please test last version we did merged a fix

closed 2025-02-09 by riderx

Albermonte, 2025-02-09:
It worked, thank you.
I would have waited for `isInitializing` to be false instead of returning because right now the `stop` function will return an error and don't stop the camera at all, but that's also fine for me since I can wait for the initialization on the app side