Cap-go/capacitor-camera-preview #178: bug: y-offset not working on CameraPreviewOptions

TobiasHesselinkTopicus, 2025-02-27:
### Version

6.5.7

### Platform(s)

- [x] Android
- [x] iOS
- [ ] Web

### Current behavior

The offset for the y-axis does not work properly.
I've set for this purpose the offset to 106px as this is the size of the header (by getting the height of the ElementRef).

I also placed an absolute `<div>Test</div>` at y-position 106px to mark the exact position.

See the provided screenshots which show that the camera preview is positioned underneath the header. This is easy to see in the second screenshot where i made the background of the header transparant. It looks like the provided `y` value is not the same as the amount of pixels used in the Ionic application.

This is tested on a iOS device (iPhone 16 Pro).

**This is how i start the CameraPreview:**
```
@ViewChild('cameraPreview', { read: ElementRef })
public cameraPreviewRef: ElementRef;
...
const cameraPreviewHeight = this.cameraPreviewRef.nativeElement.offsetHeight;
const cameraPreviewOptions: CameraPreviewOptions = {
    position: 'rear',
    parent: 'cameraPreview',
    toBack: true,
    disableAudio: true,
    y: 106,
    height: cameraPreviewHeight
};
return from(CameraPreview.start(cameraPreviewOptions));
```

**This is my html which renders the CameraPreview:**
```
<ion-content
    #cameraPreview
    id="cameraPreview"
> .... </ion-content>
```

**Screenshots:**
[image]
[image]

### Expected behavior

y-axis is the same as the pixels on the screen. Now the preview can't be used properly due to incorrect sizes.

### Steps to reproduce

See code in description.

### Capacitor doctor

💊   Capacitor Doctor  💊 

Latest Dependencies:

  @capacitor/cli: 7.0.1
  @capacitor/core: 7.0.1
  @capacitor/android: 7.0.1
  @capacitor/ios: 7.0.1

Installed Dependencies:

  @capacitor/cli: 6.1.2
  @capacitor/core: 6.1.2
  @capacitor/android: 6.1.2
  @capacitor/ios: 6.1.2

### Before submitting

- [x] I have read and followed the [bug report guidelines](https://capawesome.io/contributing/bug-reports/).
- [x] I have attached links to possibly related issues and discussions.
- [x] I understand that incomplete issues (e.g. without reproduction) are closed.

### Fastest way to fix this issue

- [ ] I want to get this fix ASAP and [found the issue with Algora](https://console.algora.io/org/gl/bounties/community?fund=Cap-go%2Fcapacitor-inappbrowser) we recommend ($50 to $500)

TobiasHesselinkTopicus, 2025-03-06:
Any help on this would be appreciated. Thanks in advance. @riderx

TobiasHesselinkTopicus, 2025-04-02:
@riderx Still waiting for your response. Any update so far? Would be very much appreciated.
Thanks in advance.

riderx, 2025-04-02:
Sorry i totally missed it ! I'm currently in vacation but @WcaleNieWolny can have a look

WcaleNieWolny, 2025-04-03:
I can try to fix this. An ETA is tomorrow morning at the latest

WcaleNieWolny, 2025-04-03:
@TobiasHesselinkTopicus question: Is this an Android-only issue? From what you shared, it looks like iOS works well. Or perhaps I am interpreting what you wrote wrong

WcaleNieWolny, 2025-04-04:
Hi, please update to the latest version (preferably V7, but if not possible, the latest V6). Also, please provide a detailed list of ALL the plugins that you use. Do you use Capacitor `StatusBar`? I wasn't able to reproduce it in the latest release from main.

I will be closing this issue. Please provide more detailed reproduction steps and/or a reproduction app. I am not able to reproduce.

TobiasHesselinkTopicus, 2025-04-10:
Dear @WcaleNieWolny @riderx,
I've created a demo app which you can use. Please see the README inside for more details about the issue.
Hope you can take a look at it. Would be very much appreciated if we can solve this problem.
Thanks in advance.
[demo-app.zip](https://github.com/user-attachments/files/19685875/demo-app.zip)

WcaleNieWolny, 2025-04-12:
I still don't understand this bug.

WcaleNieWolny, 2025-04-12:
This might be an iPhone only issue... 

[image]

I did modify your demo app slightly to do
```
    const cameraPreviewOptions: CameraPreviewOptions = {
      position: 'rear',
      parent: 'cameraPreview',
      toBack: true,
      disableAudio: true,
      y: 106, //offsetY,
      height: cameraPreviewHeight - (106 - headerHeight),
    };
```

but otherwise, I still don't understand the issue. Also, your demo doesn't work for landscape orientation

riderx, 2025-04-12:
Hey @TobiasHesselinkTopicus i added option `includeSafeAreaInsets` who should fix you issue i believe have a try

TobiasHesselinkTopicus, 2025-04-14:
@riderx Thanks for the possible fix. I want to try if this fixes the issue, but i see the Package release action is broken.
See your the 7.2.0 release build: https://github.com/Cap-go/capacitor-camera-preview/actions/runs/14422014664/job/40445781576
Any chance to fix this and republish the new version?

WcaleNieWolny, 2025-04-14:
@TobiasHesselinkTopicus I just pushed https://github.com/Cap-go/capacitor-camera-preview/commit/1479e790f5854d173c66e4a1d35cee65a5248ba3. It should fix the issue

WcaleNieWolny, 2025-04-14:
I have no idea why NPM has been rejecting our automated builds recently. I saw the same behavior in the StreamCall plugin. 

CC: @riderx

WcaleNieWolny, 2025-04-14:
The change is live in the '7.2.2' version. Sorry for this wired build bug

TobiasHesselinkTopicus, 2025-04-14:
@riderx @WcaleNieWolny Thanks for the deploy. Just tested the new `ignoreSafeareaInsets` property, but unfortunately this does not work. When setting this to true, the `y` offset is positioned far worse 👀 . Check my screenshot for the result.

[image]

TobiasHesselinkTopicus, 2025-04-14:
Also good to know, checking safeareaInsets not on the main Thread causes issues:
[image]

riderx, 2025-05-07:
Hello @TobiasHesselinkTopicus we currently busy on other task and don’t have time to check before a month at least.
If you want it fast you can sill buy priority support and we will fix it quickly.
You can book a call here https://cal.com/team/capgo/meet if you want to know more about premium support

closed 2025-08-21 by riderx