Cap-go/capacitor-camera-preview #200: feat: Detect Face and face position to show some filters

Saqib92, 2025-07-08:
### Current problem

Can we Add Detect Face and face position to show some filters like snapchat. 

### Preferred solution

Should expose realtime face position

### Alternative options

Alternative in React Native:
https://github.com/luicfrr/react-native-vision-camera-face-detector

### Before submitting

- [x] I have read and followed the [feature request guidelines](https://capawesome.io/contributing/feature-requests/).
- [x] I have attached links to possibly related issues and discussions.

### Fastest way to get this feature added

- [ ] I want to get this feature ASAP and [found the issue with Algora](https://console.algora.io/org/gl/bounties/community?fund=Cap-go%2Fcapacitor-inappbrowser) we recommend ($100 to $1000)

riderx, 2026-01-30:
@Saqib92 we just closed 2 PR because it look unfnished fr us but maybe you wanna test it locally and see if that help you ?

mosalah12272-cyber, 2026-02-21:
@algora-bot I have submitted the Android implementation in PR #313. Please check