Cap-go/capacitor-camera-preview #216: bug: Android: Camera not showing

graphefruit, 2025-08-28:
### Version

7.6.1-alpha.3

### Platform(s)

- [x] Android
- [ ] iOS
- [ ] Web

### Current behavior

On Android the camera does not show up.

You'll see a screenshot here:

[image]

### Expected behavior

Camera on Android shows up.

### Reproduction

https://github.com/graphefruit/Beanconqueror/

### Steps to reproduce

I could write a sample APK for you to install if needed.
Feature branch: camera-preview

### Other information

What I'm trying to achieve:
I'm having a live graph for pressure and weight data from bluetooth devices for a coffee extraction. 
The camera shall display the portafilter, to have a look at the graph and at the same time also the extraction phase on the portafilter.

What I've noticed:
I see it very shortly flashing when starting the camera, and then nothing.
I've tried it on Android 11 and Android 14 (Pixel 2XL, Pixel 4A5G)

Adding to this I tried it with the https://github.com/capacitor-community/camera-preview plugin. 
Here the camera on iOS and Android works, but when using on iOS, the pixel-ratio is differently and the x/y setting is not working correctly.

Also I've played a lot with different settings, with front/rear camera, deleting the whole dom, so just the DIV with camera preview was showing, adding everywhere `background-color:transparent` without success

I had before 7.6.0 and tried it also with the current alpha but that didn't fix it.

The coding lines can be found here:
https://github.com/graphefruit/Beanconqueror/blob/241c8a46837049b4afae980e08ed21692e6f38a7/src/app/brew/brew-flow/brew-flow.component.ts#L503

Thank you so much for this wonderful plugin and the time and effort put into this!
Have a great cup of coffee
Lars

### Capacitor doctor

 npm run cap doctor

> beanconqueror@2.0.0 cap
> capacitor doctor

💊   Capacitor Doctor  💊 

Latest Dependencies:

  @capacitor/cli: 7.4.3
  @capacitor/core: 7.4.3
  @capacitor/android: 7.4.3
  @capacitor/ios: 7.4.3

Installed Dependencies:

  @capacitor/cli: 7.4.2
  @capacitor/core: 7.4.2
  @capacitor/ios: 7.4.2
  @capacitor/android: 7.4.2

[success] iOS looking great! 👌
[success] Android looking great! 👌

### Before submitting

- [x] I have read and followed the [bug report guidelines](https://capawesome.io/contributing/bug-reports/).
- [x] I have attached links to possibly related issues and discussions.
- [x] I understand that incomplete issues (e.g. without reproduction) are closed.

### Fastest way to fix this issue

- [ ] I want to get this fix ASAP and [found the issue with Algora](https://console.algora.io/org/gl/bounties/community?fund=Cap-go%2Fcapacitor-inappbrowser) we recommend ($50 to $500)

WcaleNieWolny, 2025-09-01:
Please make absolutely sure your web view is transparent, otherwise you will not be able to see the preview. You can also set `toBack` to `false`. This will place the preview ON TOP of you webview, but you will not be able to place UI elements on top of your camera preview

WcaleNieWolny, 2025-09-02:
Hi, can you update to the latest version (7.8.1). There are major changes between these versions, and I cannot really provide support for the ancient/pre-rewrite version of the preview.

WcaleNieWolny, 2025-09-02:
I cloned your repo, so if this does nothing to help I will have a look myself

graphefruit, 2025-09-02:
> I don't understand your app. I am sorry, but I read the code, removed the `isIos` check from the camera preview so it's visible on iOS and I still don't see the button. Please help me figure this out so that I can help you

You sadly need bluetooth devices to make this work.
I'll do a test-branch for you to work without bluetooth devices.
I'll commit this evening and let you know.

But first I'll update to the latest version.

Thank you so much for your fast response and your help here! Realy appreciated.

Have a great cup of coffee
Lars

graphefruit, 2025-09-02:
Hey @WcaleNieWolny !
I've just published the branch: https://github.com/graphefruit/Beanconqueror/tree/feature/camera-preview-android

Basically when you're skipping the welcoming steps, and are on the dashboard you'll see the big red button "toggle camera".

[image]

The code is executed in the dashboard: 
https://github.com/graphefruit/Beanconqueror/blob/48708c5a4ef0c248ca17333fb42a6a89a1cb4251/src/app/dashboard/dashboard.page.ts#L58

Also I've added a transparent background to all elements I can find: 
https://github.com/graphefruit/Beanconqueror/blob/95f2d6fc770eaad37175bfe902318847c442104c/src/app/app.scss#L7

For building the whole code to deploy yourself:
```
npm run -- build && npm run capsync
```
And then you can use android-studio to generate an APK.

Let me know if I shall ship you an APK, and add debugging points to it.

Thank you again for your support!

kevinfavv, 2025-09-04:
I ran into the same problem.
It looks like the default WebView on Android is opaque.

I solved it by adding a minimal override in my MainActivity.java:

``` java
// Inside the MainActivity Class
@Override
protected void onCreate(Bundle savedInstanceState) {
    super.onCreate(savedInstanceState);

    getWindow().setBackgroundDrawable(new ColorDrawable(Color.TRANSPARENT));
    getBridge().getWebView().setBackgroundColor(Color.TRANSPARENT);
}
```

I opened a pull request to add this to the README, but I’m not sure if this behavior is present across all Android versions.
https://github.com/Cap-go/capacitor-camera-preview/pull/224

riderx, 2025-09-04:
@graphefruit that exactly the use case we test, can you define what is not possible ? do you make all modal transparent or just the top one ? Sadly we can't put the camera view between webview layer so it will always been to the bottom

graphefruit, 2025-09-06:
> It was originally a fork but this summer we did full rewrite of Android as the current one use all deprecated API and cannot support Virtual camera and all new features Android has.

Thanks! Understood, that means it won't be possible right in a future to still have it possible to overlay?

riderx, 2025-09-07:
Not sure what you mean, we did full rewrite to make it work because the old was had thousands of bugs.
So our goal is to make it work without bug, on back and front mode. Back mode works in our test so we just need to find why for your case it doesn’t. I started this week to implement what @kevinfavv shared but got sick so i left it on the side.
You tried his fix does it work for you or not?
Did you made all sub layers of your app transparent ?

kevinfavv, 2025-09-07:
> Thank you [@kevinfavv](https://github.com/kevinfavv) I tried this and it definitily works. **But** what I want to achieve: I have multiple modals stacking and on the last modal on front I want to display the camera in a specific area, which is basically not possible even with this change.
> 
> What I don't quite understand why the capacitor-camera is working in this with Android.

The fact that it works inside a modal is not an issue related to the plugin.
In my codebase, I made it work inside an Ionic modal.

To do this, I simply created a modal without a backdrop, and when opening it, I set the background color variable to transparent.
On top of that, I set the main ion-router-outlet (selected here with #router) to opacity = 0.

(My exemple is on vuejs but same logic apply on other frameworks)

**App.vue**
```html
<template>
  <ion-app>
    <ion-router-outlet id="router" />
  </ion-app>
</template>
```

**ModalService.ts** – My app code for opening a modal with the modalController
```javascript
export async function openCameraUiModal(conversationId: number) {
  document.documentElement.style.setProperty('--ion-background-color', 'transparent', 'important');
  const router = document.getElementById('router');
  if (router) router.style.opacity = '0';

  await modalController.create({
    id: "modal-name",
    component: ModalComponent,
    componentProps: { props1: true, x: 1, y: 1 },
    animated: false,
    showBackdrop: false
  }).then(m => m.present());
}

```

This code instantly creates a transparent modal and hides the content underneath it.

Of course, you need to reset the app back to its normal state. To do that, in the onUnmounted() hook of the modal, I added this:
```javascript
onUnmounted(() => {

  document.documentElement.style.setProperty('--ion-background-color', '#fff', 'important');
  const router = document.getElementById('router');
  if (router) router.style.opacity = '1';

});
```

This little hack has nothing to do with a potential bug in the plugin, it's just my workaround that might help you unblock your situation.

riderx, 2025-09-09:
In our example we make a service and we include it in every page who is in the stack view to make it transparent, that the only way to my knowledge to make it work everywhere

closed 2025-09-09 by riderx

Asko-Dev, 2026-01-30:
still happening, my client only disclosed xiaomi redmi, photos are taken, preview not visible