Cap-go/capacitor-camera-preview #231: bug: Android: Camera not showing

elov4anin, 2025-09-13:
### Version

7.14.5

### Platform(s)

- [x] Android
- [ ] iOS
- [ ] Web

### Current behavior

"@capacitor/android": "^7.4.3", "@capacitor/cli": "^7.2.0", "@capacitor/ios":"^7.4.3","@capgo/camera-preview": "^с",
The bug repeats on Xiaomi Redme Note Pro, Android 11, MIUI 12.5.3, the methods described in the topic do not help. There is no camera preview, but pictures are being taken
I also installed the application from the example - the camera preview is also not shown.

### Expected behavior

Camera on Android shows up.

### Steps to reproduce

This bug is the same as in the issue https://github.com/Cap-go/capacitor-camera-preview/issues/216 .
I used all the solutions listed there.

[image]

### Capacitor doctor

💊   Capacitor Doctor  💊 

Latest Dependencies:

  @capacitor/cli: 7.4.3
  @capacitor/core: 7.4.3
  @capacitor/android: 7.4.3
  @capacitor/ios: 7.4.3

Installed Dependencies:

  @capacitor/cli: 7.4.3
  @capacitor/core: 7.4.3
  @capacitor/ios: 7.4.3
  @capacitor/android: 7.4.3

### Before submitting

- [x] I have read and followed the [bug report guidelines](https://capawesome.io/contributing/bug-reports/).
- [x] I have attached links to possibly related issues and discussions.
- [x] I understand that incomplete issues (e.g. without reproduction) are closed.

### Fastest way to fix this issue

- [ ] I want to get this fix ASAP and [found the issue with Algora](https://console.algora.io/org/gl/bounties/community?fund=Cap-go%2Fcapacitor-inappbrowser) we recommend ($50 to $500)

riderx, 2025-09-13:
Please try latest i made a compat change for this

elov4anin, 2025-09-15:
@riderx  Thanks a lot, everything worked in 7.15.0